bambulab/BambuStudio #391: [BUG / 01.02.00.09 / macOS] Bambu Studio crashes when saving a project

Gio76, 2022-10-07:
- **Bug Description:**
When Bambu Studio is in Full Screen, it crashes every time you save a new project. After clicking on `Save Project` from the `File` menu, Bambu Studio opens the system window where you can enter the file name and choose the file location, once you click on `Save`, the app crashes.

- **Screenshot:**
  [image]

- **Bambu Studio Version:**
  v.01.02.00.09

- **Device Used:**
  [image]

- **Full Report Below:**
```
Process:               BambuStudio [21609]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.02.00.09 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-10-07 22:30:29.9465 +0700
OS Version:            macOS 12.6 (21G115)
Report Version:        12
Anonymous UUID:        264DC695-76B6-7ABF-7B68-D135BF7DCFFC

Sleep/Wake UUID:       C9F81492-21B0-4EC0-A9D2-C145DE154BDC

Time Awake Since Boot: 89000 seconds
Time Since Wake:       1948 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustudio_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000018
Exception Codes:       0x0000000000000001, 0x0000000000000018
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [21609]

VM Region Info: 0x18 is not in any region.  Bytes before following region: 105553518919656
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Thread 0 Crashed:: bambustudio_main Dispatch queue: com.apple.main-thread
0   libobjc.A.dylib               	       0x1a03c65a0 objc_msgSend + 32
1   BambuStudio                   	       0x1052c99f4 Slic3r::GUI::Plater::priv::set_project_filename(wxString const&) + 304
2   BambuStudio                   	       0x1052dd598 Slic3r::GUI::Plater::export_3mf(boost::filesystem::path const&, Slic3r::SaveStrategy, int, std::__1::function<void (int, int, int, bool&)>) + 3856
3   BambuStudio                   	       0x1052dc46c Slic3r::GUI::Plater::save_project(bool) + 216
4   BambuStudio                   	       0x106f15c2c wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
5   BambuStudio                   	       0x106f158f8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
6   BambuStudio                   	       0x106f15720 wxEvtHandler::ProcessEvent(wxEvent&) + 56
7   BambuStudio                   	       0x106f15dd8 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
8   BambuStudio                   	       0x106cf2518 wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) + 160
9   BambuStudio                   	       0x106cf23f0 wxMenuBase::SendEvent(int, int) + 152
10  BambuStudio                   	       0x106df42f0 wxMenu::HandleCommandProcess(wxMenuItem*) + 184
11  AppKit                        	       0x1a33765c8 -[NSApplication(NSResponder) sendAction:to:from:] + 460
12  AppKit                        	       0x1a346c2ec -[NSMenuItem _corePerformAction] + 444
13  AppKit                        	       0x1a346bfe0 -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:] + 100
14  AppKit                        	       0x1a34b4bd8 -[NSMenu performActionForItemAtIndex:] + 200
15  AppKit                        	       0x1a34b4af8 -[NSMenu _internalPerformActionForItemAtIndex:] + 100
16  AppKit                        	       0x1a34b4904 -[NSCarbonMenuImpl _carbonCommandProcessEvent:handlerCallRef:] + 116
17  AppKit                        	       0x1a344fe70 NSSLMMenuEventHandler + 728
18  HIToolbox                     	       0x1a92226c8 DispatchEventToHandlers(EventTargetRec*, OpaqueEventRef*, HandlerCallRec*) + 1084
[... 476 lines cut ...]
      "rolloutId" : "60da5e84ab0ca017dace9abf",
      "factorPackIds" : {

      },
      "deploymentId" : 240000008
    },
    {
      "rolloutId" : "61301e3a61217b3110231469",
      "factorPackIds" : {
        "SIRI_FIND_MY_CONFIGURATION_FILES" : "6216ae152a40e71046e16225"
      },
      "deploymentId" : 240000016
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "5b729754-3882-4560-826f-9c0bc27d77c6",
      "experimentId" : "6192fb082171a2330e561df0",
      "deploymentId" : 400000035
    }
  ]
}
}

```

bambu123, 2022-10-09:
Thank you for reporting the issue. We will fix it soon.

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

closed 2022-10-21 by YifanWuBambu