bambulab/BambuStudio #393: Z motion loud noise after print

parksddd, 2022-10-07:
On X1 non-carbon when the print completes the Z motors produce an extremely loud noise while lowering the bed.
It seems to be after an update that reduced the power to the z-motor.

Please consider increasing the power to the motor to reduce this noise.

RobinWhitfield, 2022-10-07:
This has been discussed a little bit on the Discord. It seems some specific printers slip z-steps with the new lower motor current value.

Raising it back to 0.45 or so resolves the issue.

parksddd, 2022-10-07:
Yes. It would be nice if Bambu fixed it in the code for the default profiles.

RobinWhitfield, 2022-10-07:
The presumption is that it was done in order to further reduce the potential for damage when lowering the bed, so lowering the speed the bed descends at might be a suitable alternative option to raising the current again.

ZzzGin, 2022-10-09:
@RobinWhitfield , is there a Discord channel? Is it public (for all printer purchasers)?

RobinWhitfield, 2022-10-09:
> @RobinWhitfield , is there a Discord channel? Is it public (for all printer purchasers)?
https://discord.com/invite/Jmby9N4t8E

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

closed 2022-10-21 by YifanWuBambu