bambulab/BambuStudio #395: Park extruder over filament purge after completing print

brandonraineri, 2022-10-08:
I’m thinking it would be more logical to park the extruder over the filament purge chute after a print. That way any ooze after the print or during a preheat will be captured in the chute instead of on the bed. You can even clean nozzle during preheat before starting the print as well to prevent ooze on the bed.

kyle1046, 2022-10-08:
I would also like to see it clean the nozzle after doing a flow calibration since I keep seeing fillament sticking to the side of the nozzle from the calibration

RobinWhitfield, 2022-10-09:
> I would also like to see it clean the nozzle after doing a flow calibration since I keep seeing fillament sticking to the side of the nozzle from the calibration

I think this one's a totally different issue, but I agree entirely. The pause at the end of the bottom purge line also often causes sticky filament to stick to the end of the nozzle.

In short, the starting G-code needs some work, but this is the wrong issue for it.

ChloeShang123, 2022-10-11:
Considering the printing process which will do homing at first. If the tool head parked over filament purge, the tool head will conflict with right motor at the beginning of the next printing task.

RobinWhitfield, 2022-10-11:
> 

All this would mean is that the homing procedure needs amending to move at least 20mm in the negative Y direction before moving in the positive X direction.

There's no reason that X has to be moved before Y.

kyle1046, 2022-10-11:
> > 
> 
> All this would mean is that the homing procedure needs to move at least 20mm in the negative Y direction before moving in the positive X direction.
> 
> There's no reason that X has to be moved before Y.

If their happens to be a timed wait command then it can park over the fillament purge for (x) milliseconds/seconds  after sending the cooldown command befor moving back to the current park position

whiiiskyy, 2022-10-15:
This can be fixed with custom g-code.
First of all, if the filament isn't being changed, there's absolutely no reason to purge aka waste filament in the first place.

The nozzle should never be fully heated BEFORE doing the bed calibration.

If you did change the filament and a purge is needed, it should cool the nozzle to 155c or so after the process.

I've this set on my current printer/slicer. 

- Heat the nozzle only up to 155c. If I changed the filament so 220 or, the nozzle is cooled down to 155c before continuing 
- do the 7x7 mesh bed
- park the nozzle at the purge line section
- fully heat the nozzle and start

No oozing, no bed messed up.

RobinWhitfield, 2022-10-15:
Due to the bed homing and bed levelling using the end of the nozzle, heating it up to ensure it can be wiped and smushed flat on the tab at the back of the bed before ABL is important.

I agree on the purging though, and I'd like see some more smarts on the printer/printing options (like cal and ABL have) for this.

The nozzle load line should be more than sufficient when you've not changed filament.

Personally I have a couple of profiles that I've modified where it totally cuts out all of the purge. I swap filaments and purge manually anyway as I've not got an AMS.

whiiiskyy, 2022-10-15:
> Due to the bed homing and bed levelling using the end of the nozzle, heating it up to ensure it can be wiped and smushed flat on the tab at the back of the bed before ABL is important.
>
> I agree on [...]

Same, I got X1C without AMS so that auto purge has to go.