bambulab/BambuStudio #397: [Feature Request] Looking for Profiles & Presets for 0.6mm nozzle

lord-x-type, 2022-10-08:
Hi there,
i was looking for some profiles/presets for a .6mm nozzle. As i looked in the slicer  i recognized there is only 1 Profile for .3mm layer hight.
Would it be possible to publish some more presets for like at the .4mm nozzle from draft to super fine?

This would be grate!
Best regards

YifanWuBambu, 2022-10-09:
[image]

Please update to 1.2.0.9.

closed 2022-10-09 by YifanWuBambu

lord-x-type, 2022-10-09:
Sorry for reopening.. but this did not solve the problem.
[image]
i am using the latest source, had added the 0.6mm nozzle but still have only the 0.3mm layer hight profile when selecting the .6mm nozzle preset..
[image]
what am i doing wrong?
is there a way to get other ones?