bambulab/BambuStudio #398: Allow more refined selection of support filaments

misterpeppery, 2022-10-08:
When choosing "0" as the support material it also includes the support interface material as a possible material for building supports. This is bad because the support interface material is often not melt compatible with the other print materials. An example is using Bambu Lab Support W with PLA. By combining Support W into the support structure it weakens the support structure. Please allow a selection for support material that excludes the support interface material. Also, there may be times when a specific filament is almost empty or particularly precious and you would like to exclude it from being used as support material, but you would still like the added efficiency that comes with allowing Bambu Lab to choose between the other materials for supports.

These same concerns may be valid for purge into infill, but less so. Consider adding a more refined filament selection for purge into infill as well.

YifanWuBambu, 2022-10-09:
@misterpeppery 
Thanks for reporting this issue.
We will optimize logics for the Support material.

YifanWuBambu, 2022-10-21:
Optimized in v1.3.0.12.

closed 2022-10-21 by YifanWuBambu