bambulab/BambuStudio #405: [FEATURE] Volumetric flow rate settings just for filament swap purges

Liamnerf, 2022-10-08:
When using PVA as a support material, I’ve found that the filament often jams inside the extruder due to its softness, and due to the stock volumetric flow settings causing it to try to be pushed too quickly during filament swap purges. In order to solve this, I’ve had to set the volumetric flow rate for BOTH filaments to very low values. While this solves the issue of PVA clogging, it introduces a new issue: the printing also becomes dead slow, when the filaments in question could certainly be printed faster. 

And so, my proposal: allow us to set a separate volumetric flow rate value for each filament that only applies to the purging done during filament swaps. 

By introducing this setting, it will allow print speeds to be kept up, while filament swap purges can be kept nice and slow for sensitive materials like PVA, to prevent jamming.

YifanWuBambu, 2022-10-09:
Hi @Liamnerf ,
I printed with poly dessovle and it worked fine.

Could you help provide the brand and family of your PVA filament?
With your PVA, what is the desire "Max volumetric speed" in your printing?

Liamnerf, 2022-10-09:
Hi @YifanWuBambu , 

https://3dfilaprint.com/product/1-75mm-pva-s-500gms-by-dutch-filaments/

This is the PVA I have been using.

I currently have the max volumetric flow rate for both filaments (PVA and PETG) set to 4. I have found that beyond this causes jams, even if I set the petg VFR higher but keep the PVA at 4. In every case, it is the PVA that ends up jamming in the extruder gears. 

I am fine with keeping the VFR at 4 for PVA, but with PETG I should certainly be able to print faster. I feel that adding separate VFR control for filament flushes could solve this. What are your thoughts on this?

closed 2022-10-21 by Liamnerf