bambulab/BambuStudio #415: [Feature] Ability to extract flow calibration data.

Jollzay, 2022-10-10:
I figure this has probably been mentioned, but I cannot stress how nice it would be, to read the data from flow calibration, and copy-paste it into the userpresets in the slicer. 
Having to swap build plates (using textured PEI quite alot), to get an accurate calibration it getting tedius. 
Also having this feature for multi material printing, would be a game changer. 

Perhaps even include an "AMS Flow Calibration" option, where you fill up the AMS with 4 filaments, then run a calibration program. Each filament gets a section on the build plate (there should be enough space for 4 calibration patterns.) Then is saves all the data for each slot, and you're good to go. Even having this option just 1 filament would be nice. 

Thanks

brandonraineri, 2022-10-12:
^What this guy said^

Learoy1, 2022-11-04:
We all want this, particularly having all loaded filaments in the AMS to be properly individually calibrated.

NairoDorian, 2023-01-10:
Yes yes yes we all want this, personally, this is my top 1 most wanted feature, this could be huge and gamechanger, release the data please

brandonraineri, 2023-01-10:
Bambu has done an excellent job integrating feature requested but yes I feel this one is a high priority for many people but has yet to be implemented. Hopefully in the next update.

esc247, 2023-02-12:
I cannot agree more!