bambulab/BambuStudio #417: [Bug] [Issue] Support Blocker in Paint in Support not working

YyAoMmIi, 2022-10-11:
When support painting, right click = support blocker, aka Support should not appear there. This is not functioning as intended

See the hole in the foot has paint in support blocker
[image]
Yet
Preview shows support there
[image]

Different model
https://www.printables.com/model/292343-silica-dry-box-for-bambulab-ams
[image]
See how only 1 section has support, rest has support blockers.
[image]
Using Normal(auto), clearly get support at where support blockers is.
[image]
Same issue for Hybrid(auto)
[image]
Issue for Normal, no auto. Normal to my knowledge, should only apply support where it is painted
[image]
Tree Support kinda worked, but get something weird there.

MusangKingBambulab, 2022-11-28:
Hi, @YyAoMmIi , hope you are still enjoying 3D printing. As for this issue, you mentioned last month, has been solved by adding a new config called "Support xy expansion" and it will come out with the next release. We open an extra setting for users to adjust the overhang area so that in this case, you can shrink the normal support span the way you like. For this dry box model, I only block the beneath overhang plane inside the box and set "Base pattern spacing" and "Support xy expansion" with 2mm and -1.5, respectively under Normal support mode. The result is quite close to what you want. (see below figs). We are always hearing your valuable suggestions. 
[image]

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

ptsao88, 2023-01-08:
@YifanWuBambu @MusangKingBambulab I tried this today on 1.4.0.17 and it did not work.

Can you confirm this got included into v1.4 properly?

See my comment here: https://github.com/bambulab/BambuStudio/issues/437?fbclid=IwAR38W8ZkFsNLMgG-VhANLTSN5B8kJkkUD34kqpequlTZptL7QAgovL8RICs?fbclid=IwAR38W8ZkFsNLMgG-VhANLTSN5B8kJkkUD34kqpequlTZptL7QAgovL8RICs#issuecomment-1374921395

MusangKingBambulab, 2023-01-09:
Hi, @ptsao88, we did integrate this feature into the current release, and please check below. Three parameters I changed are for your reference with this model.
[image]

ptsao88, 2023-01-09:
Hi @MusangKingBambulab - this still isn't working as I expect. See below comments.

If I import the model and there are NO support blockers painted, then I can see the Normal Support Expansion shrinking as expected.

[image]
[image]
[image]
[image]
[image]

However, once I paint blockers/enablers, this no longer works as expected...also the support blockers are still not working properly and there are still supports being generated where the model is painted red. This is the same whether normal (auto) or normal (manual) is chosen, and whether or not the "Don't Support Bridges" is checked or not.

[image]
[image]
[image]
[image]

MusangKingBambulab, 2023-01-10:
@ptsao88 Got it. Don't worry, I will continue to follow this issue. Maybe the snug mode is alternative choice for us.

ptsao88, 2023-01-10:
@MusangKingBambulab possibly...although I would like to see the support enforcer and blocker actually work properly (eg. a blocker will prevent supports from being generated on that surface).

If there was a way to delete supports after generation, that would also be useful.

It's also confusing that changing the "Base Pattern Spacing" causes the support generation to change so much.  I would expect that "Base Pattern Spacing" should be equivalent to "Support infill" in other slicers...which would just change the infill percent in the support area. However, you can see that by changing "Base Pattern Spacing", you sometimes end up with dramatically different supports generated (eg. supports in the small bridge areas even though "Don't support bridges" is enabled.

I would recommend looking at how Ideamaker from Raise3D generates and manipulates supports. From my experience it's been the easiest and most intuitive way to generate and modify supports.

[image]
[image]
[image]
[image]

MusangKingBambulab, 2023-01-10:
@ptsao88 Thanks for your constructive suggestion. Editable support no matter what tree or normal mode will be a higher-level using experience and we will consider it, although there is still a long way to reach. Thanks again.