bambulab/BambuStudio #418: [Feature Request] Connect Flow Calibration Lines For Easier Line Removal

talldonkey, 2022-10-11:
Feature Request:  Connect all flow calibration lines with filament so it can be all removed in one long string, and only need to start/peel one line off the bed, and then the entire flow calibration pattern can be peeled off.

Currently the flow calibration pattern requires 3 distinct areas/lines to be peeled up to be removed.  

The largest pattern, at the back/topmost, several lines/areas are all connected, so if I get one line up, I can pull off the upper pattern in one long, kinked line of filament.   However, then theres the other two lines that are not connected, but could be.

Sometimes the filament is stuck really well to the bed, and it is difficult enough to remove one, and two more gets annoying, especially when printing a lot and doing flow calibration before every print.

This is more of a quality of life/convenience request.

Thank you for your consideration.

NairoDorian, 2023-01-10:
Yes, that is very annoying for some filament to take time to scrap every line, if all are connected it would make things easier

kriordan37, 2023-07-19:
+1 great idea

SaltWei, 2023-10-08:
[image]
Does current pattern improve this issue?

talldonkey, 2023-10-09:
Yes! The current pattern does fulfill this feature request.