bambulab/BambuStudio #419: BambuStudio Crash Error exc handler [3583]

Samiu1, 2022-10-11:
When opening BambuStudio, it crashes on my macbook pro. Worked prior to October 5. 

Below is the screenshot of the error messages as well as a screenshot of the specs of my macbook

[image]
[image]

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [3583]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.02.00.09 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-10-08 09:06:56.2371 -0400
OS Version:            macOS 12.6 (21G115)
Report Version:        12
Anonymous UUID:        2861EC78-9BE4-FDB2-5E3C-780FDCA2356A

Time Awake Since Boot: 3100 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustudio_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [3583]

VM Region Info: 0 is not in any region.  Bytes before following region: 4304142336
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1008c0000-1037a8000    [ 46.9M] r-x/r-x SM=COW  ...S/BambuStudio

Thread 0 Crashed:: bambustudio_main Dispatch queue: com.apple.main-thread
0   ???                           	               0x0 ???
1   BambuStudio                   	       0x1011814ac Slic3r::GUI::GLModel::init_from(indexed_triangle_set const&) + 152
2   BambuStudio                   	       0x10112c454 Slic3r::GUI::GLGizmoBase::GLGizmoBase(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int) + 348
3   BambuStudio                   	       0x10112d608 Slic3r::GUI::GLGizmoMove3D::GLGizmoMove3D(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, Slic3r::GUI::GizmoObjectManipulation*) + 36
4   BambuStudio                   	       0x1011209c4 Slic3r::GUI::GLGizmosManager::init() + 420
5   BambuStudio                   	       0x1010cc040 Slic3r::GUI::GLCanvas3D::init() + 796
6   BambuStudio                   	       0x1011e781c Slic3r::GUI::GUI_App::post_init() + 1232
7   BambuStudio                   	       0x101215024 wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::$_25>::operator()(wxEvtHandler*, wxEvent&) + 548
8   BambuStudio                   	       0x102eadc2c wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
9   BambuStudio                   	       0x102ead8f8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
10  BambuStudio                   	       0x102ead720 wxEvtHandler::ProcessEvent(wxEvent&) + 56
11  BambuStudio                   	       0x102e10e3c wxAppConsoleBase::ProcessIdle() + 92
12  BambuStudio                   	       0x102c026c8 wxAppBase::ProcessIdle() + 24
13  BambuStudio                   	       0x102d9f7dc wxApp::ProcessIdle() + 36
14  BambuStudio                   	       0x102e35f28 wxEventLoopBase::YieldFor(long) + 120
15  BambuStudio                   	       0x102bcc740 wxWebViewWebKit::RunScriptSync(wxString const&, wxString*) const + 256
16  BambuStudio                   	       0x102bccb94 wxWebViewWebKit::RunScript(wxString const&, wxString*) const + 84
17  BambuStudio                   	       0x102bcd714 wxWebViewWebKit::AddScriptMessageHandler(wxString const&) + 344
18  BambuStudio                   	       0x10106a36c wxAsyncMethodCallEventFunctor<WebView::CreateWebView(wxWindow*, wxString const&)::$_0>::Execute() + 56
19  BambuStudio                   	       0x102ead9c8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 296
20  BambuStudio                   	       0x102ead720 wxEvtHandler::ProcessEvent(wxEvent&) + 56
21  BambuStudio                   	       0x102ead570 wxEvtHandler::ProcessPendingEvents() + 236
22  BambuStudio                   	       0x102e113ac wxAppConsoleBase::ProcessPendingEvents() + 112
23  BambuStudio                   	       0x102eb697c wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 60
24  CoreFoundation                	       0x191bcc1a4 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
[... 532 lines cut ...]
  ],
  "experiments" : [
    {
      "treatmentId" : "5b729754-3882-4560-826f-9c0bc27d77c6",
      "experimentId" : "6192fb082171a2330e561df0",
      "deploymentId" : 400000034
    }
  ]
}
}

Model: MacBookPro18,1, BootROM 7459.141.1, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3456 x 2234 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5
AirPort: Wi-Fi, wl0: Apr  6 2022 05:55:54 version 20.90.45.0.8.7.118 FWID 01-e7138ff2
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
`

lanewei120, 2022-10-12:
thanks for reporting this issue
we have fixed this issue recently and will be ready in next release

Samiu1, 2022-10-19:
Any idea on when the next release will be?

lanewei120, 2022-10-20:
it will be  ready in following days

YifanWuBambu, 2022-10-21:
Fixed in v.1.3.0.12.

Samiu1, 2022-10-21:
I still have the same issue, please see the error log below. 

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               BambuStudio [66887]
Path:                  /Applications/BambuStudio.app/Contents/MacOS/BambuStudio
Identifier:            com.bambulab.bambu-studio
Version:               01.03.00.12 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2022-10-21 10:54:18.1445 -0400
OS Version:            macOS 12.6 (21G115)
Report Version:        12
Anonymous UUID:        2861EC78-9BE4-FDB2-5E3C-780FDCA2356A

Sleep/Wake UUID:       6E6855FC-967F-426F-9320-49BAA323036F

Time Awake Since Boot: 370000 seconds
Time Since Wake:       7152 seconds

System Integrity Protection: enabled

Crashed Thread:        0  bambustu_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [66887]

VM Region Info: 0 is not in any region.  Bytes before following region: 4369989632
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      10478c000-107774000    [ 47.9M] r-x/r-x SM=COW  ...S/BambuStudio

Kernel Triage:
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage
VM - pmap_enter failed with resource shortage

Thread 0 Crashed:: bambustu_main Dispatch queue: com.apple.main-thread
0   ???                           	               0x0 ???
1   BambuStudio                   	       0x10509efcc Slic3r::GUI::GLModel::init_from(indexed_triangle_set const&) + 152
2   BambuStudio                   	       0x10504a294 Slic3r::GUI::GLGizmoBase::GLGizmoBase(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int) + 348
3   BambuStudio                   	       0x10504b448 Slic3r::GUI::GLGizmoMove3D::GLGizmoMove3D(Slic3r::GUI::GLCanvas3D&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, Slic3r::GUI::GizmoObjectManipulation*) + 36
4   BambuStudio                   	       0x10503d57c Slic3r::GUI::GLGizmosManager::init() + 420
5   BambuStudio                   	       0x104fe67bc Slic3r::GUI::GLCanvas3D::init() + 796
6   BambuStudio                   	       0x105107bf4 Slic3r::GUI::GUI_App::post_init() + 1236
7   BambuStudio                   	       0x1051387ac wxEventFunctorFunctor<wxEventTypeTag<wxIdleEvent>, Slic3r::GUI::GUI_App::on_init_inner()::$_26>::operator()(wxEvtHandler*, wxEvent&) + 556
8   BambuStudio                   	       0x106e522a4 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
9   BambuStudio                   	       0x106e51f70 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
10  BambuStudio                   	       0x106e51d98 wxEvtHandler::ProcessEvent(wxEvent&) + 56
11  BambuStudio                   	       0x106db55b8 wxAppConsoleBase::ProcessIdle() + 92
12  BambuStudio                   	       0x106ba6fb8 wxAppBase::ProcessIdle() + 24
13  BambuStudio                   	       0x106d43e8c wxApp::ProcessIdle() + 36
14  BambuStudio                   	       0x106e5b010 wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 88
15  CoreFoundation                	       0x18cc5c1a4 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
16  CoreFoundation                	       0x18cc5bff4 __CFRunLoopDoObservers + 592
17  CoreFoundation                	       0x18cc5b640 __CFRunLoopRun + 1052
18  CoreFoundation                	       0x18cc5aa84 CFRunLoopRunSpecific + 600
19  HIToolbox                     	       0x19589a338 RunCurrentEventLoopInMode + 292
[... 436 lines cut ...]
    "queue" : "com.apple.main-thread"
  }
},
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5ffde50ce2aacd000d47a95f",
      "factorPackIds" : {

      },
      "deploymentId" : 240000207
    },
    {
      "rolloutId" : "607844aa04477260f58a8077",
      "factorPackIds" : {
        "SIRI_MORPHUN_ASSETS" : "6103050cbfe6dc472e1c982a"
      },
      "deploymentId" : 240000066
    }
  ],
  "experiments" : [

  ]
}
}

lanewei120, 2022-10-21:
the issue we fixed previously is a random crash on MacOS,  so it seems not the same with yours
1. is it a 100% blocking issue on your machine?
2. and do you mean, it suddenly couldn't work since 5th, Oct, with the same version 1.2.0.9?
3. could you remove the directory of /Users/USER/Library/Application Support/BambuStudio, and for a retry?

Samiu1, 2022-10-21:
#3 fixed my problem. Thank you.

closed 2023-03-02 by SaltWei