bambulab/BambuStudio #421: [BUG] Auto-orientation of flat disc not flat on build plate

Kraminator, 2022-10-11:
I imported a flat disc (stand) object and pressed Auto-orient. It looks like it would be flat in the build plate, but when I slice it, it does not take part of the disc; somehow, half of it gets pushed as 2nd layer only, which will cause a failed print unless you use support. Clicking on manual flat turning and choosing the surface, the disc orients correctly and can be printed.

I currently have 2 models that do this, so it seems the auto-orientation is not doing a flat lay.

[image]
[image]
[image]
[image]

tangketan, 2022-10-12:
Would you please share the model?

Kraminator, 2022-10-12:
Sorry, I can't due to the license - but it is both of the round bases of this download:
https://www.cgtrader.com/3d-print-models/miniatures/figurines/jack-skellington-and-zero-the-nightmare-before-christmas

ResiduaLiu, 2022-11-17:
There are some bugs in the auto orienation process. Some are already fixed. Please check it again in the next release.

SaltWei, 2023-03-02:
Please check in V1.5.0 Public beta

closed 2023-03-02 by SaltWei