bambulab/BambuStudio #422: Error with Bambu's own Texturized PEI Plate

ChrisPirillo, 2022-10-11:
[image]
[image]

kiliankoe, 2022-10-17:
This sometimes happens with the high temp plate as well, I assume the darker marker on both is hard for the toolhead camera to read.

timropp, 2022-10-17:
Mine does this depending on what side of the plate is up - I assumed it was due to that, but it could be a difference in the printing between the two sides. If I have the logo holes in the front right corner, it recognizes the plate. If I flip it over so the holes are in the front left, I get the "can't find localization marker" error.

toddanglin, 2022-11-21:
I switched to the Textured PEI sheets in my X1Cs and almost always get this error, too. It does not seem to have any effect, but it is a nuisance to clear.

WZ9V, 2022-12-04:
I get this quite often also, it is a pain.

mainmind83, 2023-02-14:
Automatic calibration for Flow rate on Texturized PEI plate is not recommended, [check considerations here.]( https://eu.store.bambulab.com/products/bambu-dual-sided-textured-pei-plate)

Bambu Studio should prompt with this option enabled before starting.

toddanglin, 2023-02-14:
For completeness of this thread, the "build plate localization" errors were mostly resolved in a December 2022 firmware update ([01.03.00.00(20221217)](https://wiki.bambulab.com/en/x1/manual/X1-X1C-AMS-firmware-release-history)). Suggest this Issue is closed (unless it's still a problem for many).

closed 2025-06-19 by ChrisPirillo