bambulab/BambuStudio #425: [Feature Request] Flow calibration from printers LCD

brandonraineri, 2022-10-12:
Add option to run just a flow calibration from the printers LCD.  Sometimes I like to run different filaments on a texture plate but have to run a flow calibration first on a smooth plate. But in order to do so, I need to print a small object. 

I would like the option to just run a flow calibration from the printer and store the values without having to send a file to it from BS.

psiberfunk, 2022-10-14:
careful:  By “flow” calibration here do you mean linear advance or flow ratio ?  As established elsewhere they are not actually what you would expect ..

brandonraineri, 2022-10-14:
Yes whatever calibration it does at the beginning of print if you leave it checked.

psiberfunk, 2022-10-14:
Well I bring this up because it would be even BETTER if you could store the values in the slicer profile and not need to run the flow calibration on the printer at all …

brandonraineri, 2022-10-14:
I also would like this lol

psiberfunk, 2022-10-14:
Turns out you can do it just fine in most slicers by setting the “linear advance” value :)

whiiiskyy, 2022-10-15:
We were discussing this on Discord, I can easily change that on my current printer/slicer, BS doesn't and somewhat limited.

If it cannot detect changes, etc, it's defaulted to 0.02

They use a weird M0 after the M900 K0.02 which we have no idea what it does so you can't just change that and hope for the best.

psiberfunk, 2022-10-15:
Yea #348 discusses how this should really be clarified .