bambulab/BambuStudio #426: Feature: Orthographic/Perspective camera setting

Duckshow, 2022-10-12:
Coming from using Cura, it seems strange to use an orthographic camera in a 3D setting. I often find myself confused about what angle I'm looking at the model from since there's no depth in the image. 

I can imagine some people have gotten used to it, so being able to choose between orthographic/perspective would be nice.

RobinWhitfield, 2022-10-12:
Perspective is the default (and only) camera option currently in Bambu Studio

[image]

Duckshow, 2022-10-13:
Ahh, you're right.. But I'm guessing then there must be some setting to the camera that makes it look more orthographic? For reference, here's a comparison between a cube measuring 1x1x10, seen from below, in the Unity game engine vs Bambu Slicer

[image]

RobinWhitfield, 2022-10-13:
PrusaSlicer also has no setting to adjust, but it seems some people have similarly been asking for such a feature:

https://github.com/prusa3d/PrusaSlicer/issues/3699