bambulab/BambuStudio #429: [Feature Request] Don't move objects using LMB in Prepare mode

Duckshow, 2022-10-13:
I often find myself inspecting objects up close in Prepare mode - but since the left mouse button is used for both orbiting the camera _and_ moving objects, I often move objects by mistake and have to undo. A simple fix would be to only allow movement using the translation widgets.

YifanWuBambu, 2022-10-15:
Hi @Duckshow ,
Many users get used to move an object by directly drag it.
If you just want to orbit, it is recommended to click a blank area and drag the mouse.

kiliankoe, 2022-10-17:
I definitely wouldn't want to lose the left-click orbit, but @Duckshow is right in that when you zoom far into an object, it can be hard to find a place to orbit. I sometimes get into such a place when trying to paint with the painting tools. Clicking the object would paint it, clicking outside of it would orbit. It would be fantastic if I could hold a modifier button (like option/alt for example) to only orbit and not move/paint the object.

YifanWuBambu, 2022-10-17:
@Duckshow @kiliankoe 
On paint tool (color/support/seam), you can press ctrl+mouse left button to orbit and ctrl+mouse right button to pan view.

kiliankoe, 2022-10-17:
@YifanWuBambu Thank you! That's great to know! It's command for fellow Mac users reading this. A similar option for normal navigation outside of the painting tools would be a great solution imho.

OmiCron07, 2022-10-19:
In other software, orbiting the view is done by middle mouse click drag and panning is done by shift + middle mouse click drag, or the other way around. For PrusaSlicer series, this is the weirdest thing to orbit with left click on empty space because you have to find and aim for an empty space that you don't have on zoomed models.

And the camera should orbit around the mouse, not the center of the plate.

YifanWuBambu, 2022-10-19:
> @YifanWuBambu Thank you! That's great to know! It's command for fellow Mac users reading this. A similar option for normal navigation outside of the painting tools would be a great solution imho.

For pan view, drag with right mouse button always work outside of the painting tool.
For orbit, Bambu Studio only supports click on a blank area and we will think about how to add a proper operation.

@OmiCron07 
Currently, ⌘+left mouse drag could orbit around the center of screen.

OmiCron07, 2022-10-19:
> @OmiCron07
> Currently, ⌘+left mouse drag could orbit around the center of screen.

Good to know, but kind of intuitive for me when you work with other 3D softwares.