bambulab/BambuStudio #432: Bambu Studio 01.02.00.09 - Crashes occasionally

djunod, 2022-10-14:
This version of Bambu Studio seems to crash a lot.  Just sitting on the Device page with video while printing.

Attached is the most recent crash report from macOS.

[BambuStudioCrash.txt](https://github.com/bambulab/BambuStudio/files/9787573/BambuStudioCrash.txt)

djunod, 2022-10-14:
Here's another that just happened.  Sitting on Devices watching a print.

[BambuStudioCrash2.txt](https://github.com/bambulab/BambuStudio/files/9788958/BambuStudioCrash2.txt)

StoneLiBambulab, 2022-11-21:
Hi @djunod,
    Whether the new version (01.03.00.25) still has this problem?

Thanks!
Stone Li

StoneLiBambulab, 2023-04-24:
It has been too long since there was any reply. Please use the new version. closed.

closed 2023-04-24 by StoneLiBambulab