bambulab/BambuStudio #434: Gmail login button broken

Benderrodrigu3z, 2022-10-14:
MacBook Pro 2012 10.15.7
Reinstall and copy of Mac to downgrade from BigSur to Catalina, personal issues

I removed the Library folder and now the plugin installed but the gmail button still doesn't do anything, my cursor changes shape when I hover over it but that is it, I can't login with my gmail account to have access to my printer

Benderrodrigu3z, 2022-10-14:
copying over the library files from the other Mac from Library->Appication Support->BambuStudio
Fixed it but this is an issue on the current build of 01.02.00.09

lanewei120, 2022-10-19:
not sure it is a Macos 10.15 or not
just tried it on macos 12.6 here

the behavior is correct, sometimes it is slow because of the networking

are other buttons(facebook/apple) correct?

lanewei120, 2022-11-01:
@Benderrodrigu3z 

it is an issue related with old macos

could you try this web url directly on your default browser?
https://bambulab.com/en/sign-in

as Bambu Studio using wxwidget's webview engine, which depends on the system's web core
we need firstly to find out it is the system's issue or the wxWidget's issue

Kratak, 2023-04-06:
This issue is still in version 1.5 and 1.5.x hotfix.
Buttons for third party logins aren't working.

I manage to solve it by installing previous working versions V1.4.2
Than login via google in my case
Than Install latest version, 1.5.x
And sesion in app was still alive <3 

I leave this solution here, but will be grate to fix this issue evntually.

My Machine and system:
16-in macbock pro 2023 m2 max
Ventura 13.2.1

lanewei120, 2023-04-07:
> This issue is still in version 1.5 and 1.5.x hotfix. Buttons for third party logins aren't working.
>
> I manage to solve it by installing previous working versions V1.4.2 Than login via google in my [...]

@Kratak 
thanks for reporting this issue

we have fixed this issue online
please have a try and let us know the results

closed 2023-04-18 by lanewei120