bambulab/BambuStudio #437: Support Generation XY Expansion Issue

ptsao88, 2022-10-15:
With default settings, Bambu studio seems to expand supports in the XY direction (see first picture).  This results in undesirable supports generation in some locations.  In this model specifically, supports that touch the build plate, and also some supports internally.

Furthermore, these additional supports that are generated from the expansion do not respect any previously defined support blocker areas.  Whether or not I paint the model with blockers does not prevent the extra supports from being created.

Setting the "Base Pattern Spacing" value to 0 eliminates this unwanted XY expansion...however, it also means that support infill becomes 100% (ie. a solid support block), which is also undesirable.

**Ideally these changes would be made to Bambu Studio:**
**- Allow the user to specify XY expansion separate from "Base Pattern Spacing" (also known as support infill %)**
**- For any expanded supports generated, respect previously defined support blocker areas**

[image]
[image]
[image]

YifanWuBambu, 2022-10-16:
@ptsao88 
Thanks for your suggestion. We have added this in our plan list.

ptsao88, 2022-10-18:
Great, thanks @YifanWuBambu !

BATEMAN87, 2022-11-11:
Bambu, please fix

YifanWuBambu, 2022-11-11:
This will be included in the next release.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

ptsao88, 2022-12-16:
Thank you for implementing this!

www.philiptsao.com

ptsao88, 2023-01-08:
@YifanWuBambu I had a chance to test this today...it doesn't look like "Normal support expansion" is working properly. Setting the number to a negative value does nothing and the supports look the same. It is not being shrunk but it IS being expanded properly.

[image]
[image]
[image]

ptsao88, 2023-03-14:
I'm still finding that support generation and expansion in Bambu Studio is severely lacking, especially for engineering parts with overhangs. The current implementation might be good for simple parts, or non-functional prints but it has issues for our work that IdeaMaker solves easily. See below.

This is with standard support settings. It creates very thin tall support structures that are very likely to fail and mess up the remaining print.
[image]

Changing the XY expansion to shrink helps a little but is not really ideal because it shrinks away too much of the supports.
[image]
[image]

Setting "Base Pattern Spacing" to 0 results in the ideal support placement. However, this also means that support infill is 100% which is very undesirable.
[image]
[image]

I would like to request the following:
- A way to generate supports with the same placement as when using "Base Pattern Spacing" = 0
- A way to modify the infill % when support placement is equivalent to "Base Pattern Spacing" = 0

MusangKingBambulab, 2023-03-15:
Hi, @ptsao88. Our current version of BambuStudio provides snug mode under Normal Support which may eliminate your concerns.
[image]

ptsao88, 2023-03-15:
Hi @MusangKingBambulab - PERFECT! This does exactly what I want, thank you so much!

MusangKingBambulab, 2023-03-16:
> Hi @MusangKingBambulab - PERFECT! This does exactly what I want, thank you so much!

Good to hear that. Enjoy!