bambulab/BambuStudio #439: [Issue - serious] Unusual surface texture on parts of prints at <0.2mm layer height.

RobinWhitfield, 2022-10-15:
When printing at layer heights smaller than 0.2mm, many users have experienced an unusual surface texture on what should be smooth walls.

As smaller layer heights are normally used to increase the visual quality of parts, this is effectively causing only larger layer heights to be usable for aesthetic parts, and the printer become unusual for the purpose that some people are buying it for.

I have also experienced potentially the same issue when printing at layer heights smaller than default on a 0.2mm nozzle.

Examples with a 0,4mm nozzle at 0.08mm layer heights:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Example with 0.2mm nozzle, 0.08 on left, 0.02 on right:
[image]

bani6809, 2022-10-22:
i ran into this problem on several other printers when printing under .1mm - i determined this is due to insufficient extruder step resolution when extruding tiny amounts of filament at extreme layer heights. you can actually see this effect if the extruder air prints - little individual dots come out, like morse code. when the extruder is running a print, it is basically just extruding periodic dots of plastic with large pauses in between instead of a smooth stream.

either finer microstepping resolution or higher gear reduction ratio is needed.

RobinWhitfield, 2022-10-22:
@bani6809, that's an interteresting idea, but if that was the case here then printing with smaller nozzles would be impossible without these effects.

The 0.2mm nozzle prints just fine at 0.2x0.08mm, which would be the same as 0.4x0.04.

The 0.2x0.08 benchy shown above has some issues, but they're totally different to the artifacting observed with 0.4mm nozzle prints at 0.08mm

On the surface issue generally however, I've personally been unable to reproduce it on 0.4x0.08mm prints, and it seems that for those who have observed it, it's more prevalent for some filaments compared to others.

bani6809, 2022-10-22:
probably related to viscosity. maybe try turning temp and speed down so it flows less freely.

try air printing and see if you observe the nice chain of periodic extruded dots.