bambulab/BambuStudio #440: Prime tower over-extrusion

illuminarti, 2022-10-16:
When printing a 4-material print with 0.12mm layers, the prime tower is being over extruded, resulting in the print head crashing into the tower after a few layers.

The gcode E-coords look to be calling for 52% more material than is needed based on the 0.5mm line width and 0.12mm layer height.

YifanWuBambu, 2022-10-16:
@illuminarti 
This bug has been fixed.
The new version will be released a few days later.

illuminarti, 2022-10-16:
Is there any sort of workaround in the meantime? This is blocking an important project for us.

YifanWuBambu, 2022-10-17:
Some walkaround methods:
1. use v1.1.1.1 and add prime tower manually
    https://github.com/bambulab/BambuStudio/releases
2. keep v1.2.0.9, but change to use 0.2mm layer height
3. keep v1.2.0.9 and 0.12 layer height, but disable timelapse feature