bambulab/BambuStudio #441: [bug] Closing the popup about "failed to pull filament out of extruder" leaves you with no way to resume the print.

olithraz, 2022-10-16:
Had an error message, something about unable to pull the filament back, check the extruder. I tapped close thinking I would be able to jog the head over to the front since it was only on layer 5 and I couldnt get my hands to the head to check it. 

I tapped the close button and when I was ready to resume the print, I realized that the retry button was no longer available on the lcd or on the bambu pc software. I tried pause/play, but my only option was to stop the print and start over. Thankfully it was still early so I did not waste too much plastic, but once you hit close on that you can never resume. I imagine it might happen with other error pauses.

aquatacizem, 2022-10-16:
Hi. If I remember correctly, the way to resume from this situation is to go to the settings tab from navigation menu on the left (under the home icon) and press 'Retry'. It's not terribly clear to go and do this after closing those error pop-ups, so some sort of guidance or prompt would definitely be helpful.

StoneLiBambulab, 2022-11-02:
Hi.
refs to https://wiki.bambulab.com/en/software/bambu-studio/use-ams-on-bambu-studio

just press 'Retry' on studio or screen in AMS control until the filament is pulled back.
[image]

Stone Li

YifanWuBambu, 2022-12-16:
This is fixed in v1.4.

closed 2022-12-16 by YifanWuBambu

mechengineermike, 2023-07-19:
There needs to be a CANCEL option from that menu as well. When retry algorithm also fails then you are stuck without any recourse but to try the same thing again. I think that algorithm is flawed and results in the filament geting bound in the system & requiring manual extraction. If control of the printer was permitted here (& not blocked by this popup) then I would attempt to move the print head, adjust the extruder temp, & feed filament briefly in an attempt to create enough slack to extract the filament.