bambulab/BambuStudio #442: Modifiers in Bambu Studio

RonenGru, 2022-10-16:
I was trying to slice an object using a modifier.  My aim was to use the modifier to prevent printing the perimeter of a vase and to show only the underlying infill pattern.  This use of modifiers is possible in Prusaslicer but from what I can tell is not possible in Bambu Studio.  In Bambu Studio the minimum perimeter value of a wall allowed is 1 but I want it to be 0 so that only the infill pattern shows.  For me not being able to achieve this is a problem given the effect I'm looking to create with the model.

Is there a reason why this effect is missing?  Is it there but hidden?  Any help would be appreciated.

Here are a couple of links to Prusaslicer that illustrate what I'm talking about:

https://help.prusa3d.com/article/modifiers_1767

https://www.youtube.com/watch?time_continue=1&v=6PVeh43Or-g&feature=emb_logo

Thanks,

Ronen

YifanWuBambu, 2022-10-16:
@RonenGru 
The limitation will be removed in the next release.

YifanWuBambu, 2022-10-21:
Added in v1.3.0.12.

closed 2022-10-21 by YifanWuBambu

RonenGru, 2022-10-21:
Thank you!  It's much appreciated.  RG