bambulab/BambuStudio #448: Android Bambu Handy Video Not Working (Celluar Data/ 4G)

sickerin, 2022-10-18:
When using **celluar data/ 4G** video playback doesn't seem to be working for bambu handy app. It keeps loading forever. I can control ams/ temperature etc. Video works when using wifi.
Device: Android 11 (Nothing Phone).
App Version: 1.0.5

Can devs or someone else do a quick test?

Allram, 2022-10-18:
Works fine for me with Nothing phone both on Wifi and cellular/4G/5G
Android 12.
Nothing OS 1.1.4
Bambu Handy 1.0.5

pepsiliu, 2022-10-19:
> When using **celluar data/ 4G** video playback doesn't seem to be working for bambu handy app. It keeps loading forever. I can control ams/ temperature etc. Video works when using wifi. Device: Android 11 (Nothing Phone). App Version: 1.0.5
>
It's a knowed bug on the Printer. Reboot the printer for a chance to recover it, if your Wi-Fi works fine.

SaltWei, 2023-01-09:
Since this is not slicer issue. Close
https://github.com/bambulab/Bambu-Handy is used to report app issues.

closed 2023-01-09 by SaltWei