bambulab/BambuStudio #449: AMS1 Assist Motor is Overloaded

SwagDad99, 2022-10-18:
I randomly get this error on the screen during a print.  Usually during a color change.  The filament is not tangled, and I tried a different spool.
[image]

BeGoneBaizuo, 2022-10-21:
I just got the same issue after my second print. 
[image]
[image]

SwagDad99, 2022-10-23:
Nobody cares about us...

randyre, 2022-10-24:
I get the same error even without color change. But the funny thing is that the filament has retracted on the spool as it it was doing a color change then the error happens. Hit retry and it continues just fine for a while then randomly the error re-appears. Mainly on longer prints.