bambulab/BambuStudio #451: Ludicrous Speed help

Benderrodrigu3z, 2022-10-18:
I found that I can run the printer on Polymaker ASA in Ludicrous mode at a temp of 286.
I need to profile this so I'm not having to hit Ludicrous every time

so far I've changed the following but my 7 hour print time holds to be the same.
All with a 1.66x multiplier

Outer Wall from 200 to 332
Inner Wall from 300 to 498
Sparse Infill from 270 to 448
Internal Solid Infill from 250 to 415
Gap Infill from 250 to 415
Support from 150 to 249
Travel from 480 to 797

what else do I need to increase to get the Ludicrous profile onto my screen, if I make these changes do I need to be on normal 100% mode on the printer? that is what I'm looking to do

Benderrodrigu3z, 2022-10-18:
unless there is a slicer option that just changes all this to ludicrous mode on push to printer?

SaltWei, 2022-10-19:
Need printer developers to give response about how the multiplier works.

Studio has no option to slice directly with these speeding up modes. And have no plan to add this feature at the moment.

Hercules117, 2023-01-08:
@SaltWei Having the ability to select the speed setting in the slicer before sending it to the printer would be nice, then we could change the speed and temp accordingly rather than having to change it after it starts printing manually.