bambulab/BambuStudio #452: Why can't you change the Price tag for the System Profile to make it the default for that filament type

Benderrodrigu3z, 2022-10-18:
[image]

YifanWuBambu, 2022-10-19:
The system profiles are designed as read-only, strictly provided by Bambu Lab officially.
If you want to change it, it is suggested to rename it and save as a user preset.

RobinWhitfield, 2022-10-20:
@YifanWuBambu, should this not be closed as #wontfix if the answer is that it's not going to change?

Might I suggest as a compromise for the built in filament profiles that they have zero cost associated with them? At least then it's not giving a somewhat incorrect value when slicing, rather no value at all.

Benderrodrigu3z, 2022-10-20:
That’s probably a better solution. A 0 cost I mean

Sincerely,
Shane F

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

invisibleninja06, 2023-07-06:
What exactly was included in 1.4? 

I still see costs for say the Bambu filaments which is not accurate and only way around this currently is to create custom user profiles of each Bambu filament, seem incredibly tedious. 

Would be really nice to have a user override for the Bambu filaments.

Pretty much every other parameter I can get behind having read only but the price varies by country so unless its made to update to match the cost by country (which would be neat but probably a major pain) an override for price would suffice.

Benderrodrigu3z, 2023-07-06:
You can manually edit it if you go find the plist file on Mac at least, but then the next update overwrites itSincerely,Shane FSent from my iPhone13

programbo, 2023-07-09:
Auto-updating prices for filament price changes in the user's region would be fantastic, but allowing Bambu filaments to have user editable prices would be a reasonable alternative.

Benderrodrigu3z, 2024-01-24:
> The price should not be bound to the filament default profile. Even not on Bambu filament. Since it varies from where you buy it.
> Secondly it does not take into consideration of where you live. So for me with SEK it shows a cost of 20!?  I guess thats Eur or USD. I would though be glad to buy filament to that price :)
> So i would recomend to remove the price from the locked default profile and make that user editable. Would be nice instead of having to maintain or copy/build your own.

Just create your own profile and change the numberSincerely,Shane FSent from my iPhone13