bambulab/BambuStudio #454: [Issue] Textured PEI Plate Bed Type is using the values of High Temp Plate instead

YyAoMmIi, 2022-10-18:
[image]
I had set the bed type to textured PEI, and added my own values to textured PEI [60 1st layer, 50 other layers]

However, when printing, the bed goes to 45C

this lead me to believe that while textured PEI is selected, it uses the values of high temp plate instead.

lanewei120, 2022-10-21:
Could you try the newest version?

we fixed a issue about it

YyAoMmIi, 2022-10-21:
I just started a print where textured was 65, high temp is 55. plate
selected is textured. resulting 1st layer bed temp is 55.

YyAoMmIi, 2022-10-26:
> I just double confirmed on version v1.3
> and found it works
> previously we have a issue which using wrong bed type('curr_bed_type')
> and we fixed it in v1.3
>
> could you show us where you checking the 1st layer bed temp
> and upload the wrong gcode&&3mf here?
>
> thanks

Hum, it is working now, so i guess issue is fixed.

lanewei120, 2022-10-28:
thanks, So I close this issue~

closed 2022-10-28 by lanewei120