bambulab/BambuStudio #455: Filament change in slicer, no opportunity to select slots in AMS

coat64, 2022-10-18:
When you do a filament-change in the vertical slider of the slicer, you can only select 1 slot in the AMS before print. Not a big problem but inconsistent with other multicolor prints. version 1.02.00.09

YifanWuBambu, 2022-10-19:
Hi @coat64 ,
This is the filament index in current project, not the AMS slot.
[image]

It will finally be mapped to AMS slot when sending a print job.
[image]

coat64, 2022-10-23:
You are right, it works. No idea what I was doing when posting. I had multiple filaments loaded in the slicer and in the AMS. Maybe the not enough corresponding types.

YifanWuBambu, 2022-10-24:
@coat64 
Okay. I will close the ticket first.
If you meet other problem, please create a new one ：）

closed 2022-10-24 by YifanWuBambu