bambulab/BambuStudio #457: Paint Tool Layers

randyre, 2022-10-19:
As a basic feature it should automatically color the number of top layers or walls of a given print.  Currently if a print has say 3 layers defined as a top layer or three walls only the topmost layer or outermost wall is colored.

Or maybe we need to be able to specify how many layers deep the paint tool goes. When printing only the top or outside this does not always completely cover the color underneath. 

[image]

YifanWuBambu, 2022-10-19:
@randyre 
It was changed to propogate to all top/bottom shell layers and will be included in the next release.

randyre, 2022-10-19:
I would like to salvage this print. Is there a way to just print a couple more layers of the red and blue before I remove it? I tried to make a .04 high version and set it to start at 5mm which is the top of the print but the object keeps getting reset to laying on the build plate. I want to start a print as some other number of Z besides zero but can't figure out how.

YifanWuBambu, 2022-10-19:
@randyre 
Currently, we do not have such a feature.

Britford, 2024-08-19:
@YifanWuBambu Any news on the ability to select the depth of color?  This would make the AMS much more useful.