bambulab/BambuStudio #460: Flushed filament estimate inaccurate if flushing volume set to zero

ptsao88, 2022-10-19:
Hi,

In the case where we are using the AMS with soluble support material as opposed to color printing, we generally do not care about flushing/priming volumes for color changeover.  Some soluble support material is 4X the cost of other filaments, so reducing waste here is useful, especially when there are many filament changes.

When the "Flushing Volumes" are set to zero it seems like Bambu Studio and the X1 will still do a minimum flush/purge that consumes whatever leftover filament was in the extruder during a filament cut/switch.  In fact, the minimum flush/purge seems to be slightly more than the length of the leftover cut filament which ends up working great for minimizing soluble support waste but still allowing filament changeover.

However, when the "Flushing Volumes" are set to zero, Bambu Studio also estimates the "Flushed Filament" as 0 grams, which technically is inaccurate because it does still purge the remaining length of cut filament for every swap.

It would be great if the following was implemented:

- The flushing volume parameters had an absolute minimum input value which equaled the volume of leftover cut filament (I presume this will allow Bambu Studio to calculate the "Flushed Filament" value properly
- Bambu Studio calculated the estimated "Flushed Filament" value based on the minimum purge volume if the value in "Flushed Volumes" is less than the minimum internal volume
- It would also be nice to be able to save preset Flushing Volumes into Bambu Studio so that one doesn't have to change these values every time they start a new project

[image]
[image]

YifanWuBambu, 2022-10-20:
@ptsao88 
Thanks for the sugguestions.

The flushing volume parameters had an absolute minimum input value which equaled the volume of leftover cut filament (I presume this will allow Bambu Studio to calculate the "Flushed Filament" value properly
>> We will do it.

Bambu Studio calculated the estimated "Flushed Filament" value based on the minimum purge volume if the value in "Flushed Volumes" is less than the minimum internal volume
>> The previous modification could solve this.

It would also be nice to be able to save preset Flushing Volumes into Bambu Studio so that one doesn't have to change these values every time they start a new project
>> Yes. It is in our plan and will be in recent release.

YifanWuBambu, 2022-10-20:
In our new version, each flushed filament volumes are seperated. It will release soon.
[image]

ptsao88, 2022-10-20:
Hi @YifanWuBambu.  Do you mean in v1.03?  I installed this morning to check.  See my observations below.

- I love that it now shows individual flushed filament volumes, great work!
- In the first image, do these flushed volumes include the remaining cut filament purge?  Or is this just the volume in the prime tower?
- For example, if I remove the prime tower, it no longer displays any flushed filament, which theoretically is incorrect because some is flushed/purged every time a filament change occurs (see second image).

[image]
[image]

YifanWuBambu, 2022-10-21:
@ptsao88 

1. flushed volumes include the remaining cut filament. Prime tower consumption is included in "Color print".
In the next release, we plan to separate the prime tower consumption from the color print.

2. For example, if I remove the prime tower, it no longer displays any flushed filament, which theoretically is incorrect because some is flushed/purged every time a filament change occurs (see second image).
>>I did not reproduce this issue.
[image]

ptsao88, 2022-10-21:
Hi @YifanWuBambu.  I can re-create the issue - see description below.

- If flushing volumes are default or auto-calculated, this works as expected (pictures 1 & 2)
- If flushing volumes are set to zero, this does not work even though the machine will purge the cut filament by default (pictures 3 & 4)

[image]
[image]
[image]
[image]

YifanWuBambu, 2022-10-22:
@ptsao88 
Enable the wipe tower and it will be correct.
Currently, the minimum flush logic is implemented along with wipe tower.

We need to refine this to be an independent logic.

ptsao88, 2022-10-22:
Hi @YifanWuBambu,

Ah OK - I forgot my wipe tower was disabled.

This will work.  I find that the print quality is better with the wipe tower anyway, and you have implemented the wipe tower well.  Thank you!!

closed 2022-10-26 by ptsao88