bambulab/BambuStudio #462: Bed Print Fails

Benderrodrigu3z, 2022-10-19:
There is an option in Klipper to skip a part while the machine is in motion for if an item has detached from the bed.
In this instance say I'm printing 5 objects and object 3 has detached from the bed so its printing that part in mid air then moving to another part bringing some of that filament over to the next part cause its on the nozzle. We need a way to say hey stop printing part 3 but continue printing the rest. You could place it on the print interface of the screen, where it does a preview for each said part in there respective location on the bed in a 3d view or top down view from the 3mf render, it places numbers in or next to each part and we hit a button that stops that part from printing. You would need a web interface mobile for this as well so that we can see it and tell that part to cancel via the app.

YifanWuBambu, 2022-10-21:
Yes. This feature in in out plan list, although may be not in recent release.

closed 2022-10-21 by Benderrodrigu3z

SaltWei, 2023-08-02:
This feature has been included in latest V01.06.00.00 X1 series firmware and [v01.07.01.62 studio](https://github.com/bambulab/BambuStudio/releases/tag/v01.07.01.62). And please make sure both firmware and studio is updated.

Currently it only work in one filament per plate situation, and doesn't support multi filament because of the complexity of flushing and changing filament. One filament per plate situation can still benefit a lot when has some objects failed to be printed and need to be skipped.

Thank you every one for this good suggestion!