bambulab/BambuStudio #463: Waisted movement on Color Change

Benderrodrigu3z, 2022-10-19:
On color Change why does the nozzle go to the purge wipe then move to the front left then back to the purge wipe location?
Why is this adding extra print time for no reason?

Benderrodrigu3z, 2022-10-19:
after wiping it goes to 0,0 and then back to purge bucket, why not just go -10 sliding around the wipe then purge

RobinWhitfield, 2022-10-20:
The front left location is where the filament cutter is triggered.

There is no way to avoid this movement when changing filaments.

YifanWuBambu, 2022-10-21:
The first movement: the nozzle go to the purge wipe position
We are trying to do some optimization here.

The second movement: the nozzle go to the front left
Just as @RobinWhitfield said, it is to cut the old filament before filament change. The "hit block" that helps cutting is laied at the left front corner of printer, so the tool head should go there to cut filament.

The third movement: back to the purge wipe position
This is easy to understand.

closed 2022-10-21 by Benderrodrigu3z