bambulab/BambuStudio #465: Even/Odd settings

randyre, 2022-10-19:
Trying to print a 3DLabs radio control plane, and the slicer just won't recognize most of the internal structure, it hollows out the object. PrusiaSlicer had a setting in the perimeter tab that is called even/odd to fix this. Can we look at implementing it in BambuStudio?

These are the recommended settings for PrusiaSlicer
**Layers and Perimeters**
Bottom solid layers: 0
Top solid layers: 0
External perimeters first: true
Layer height: 0.25 mm (0.30 mm for LW)
Perimeters: 1
Thick bridges: true
Perimeter Generator: Classic (not Arachne) since v2.5
 

**Advanced**
Default extrusion width: 0.42 mm
First layer: 0.44 mm
Perimeters: 0
External perimeters: 0
Bridge flow ratio: 0.8
Infill/perimeters overlap: 0%
Slice gap closing radius: 0
Slicing Mode: Even/Odd

randyre, 2022-10-19:
Merge this request with https://github.com/bambulab/BambuStudio/issues/92#issue-1327280755

SaltWei, 2022-10-21:
Could you share one 3mf of prusa project for us to compare the slicing result?

randyre, 2022-11-13:
https://3dlabprint.com/faq/prusaslicer/
[w1.3mf.zip](https://github.com/bambulab/BambuStudio/files/9998076/w1.3mf.zip)
[2115282_Stearman_PT17_test_parts.zip](https://github.com/bambulab/BambuStudio/files/9998077/2115282_Stearman_PT17_test_parts.zip)

randyre, 2022-11-14:
Here's a test piece provided by the designer.
[PolyLight LW-PLA Generic 200mm.zip](https://github.com/bambulab/BambuStudio/files/10007702/PolyLight.LW-PLA.Generic.200mm.zip)

randyre, 2022-12-10:
@SaltWei Any updates on this?

SaltWei, 2022-12-10:
Already added in internal version. Will be published soon! Thank you very much for suggestion!

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu