bambulab/BambuStudio #466: Toggle Arachne slicing engine

randyre, 2022-10-19:
Allow Arachne slicing engine to be turned on and off.

RobinWhitfield, 2022-10-20:
Right now the Arachne slicing engine is only used for internal gap filling. Is this what you're looking to be able to turn on and off, or is this also another request for #54 along with a toggle?

randyre, 2022-10-20:
Yes looks like we can merge these requests.  Support for the full engine as well as the ability to completly turn it off.

SaltWei, 2022-10-21:
Arachne has confliction with overhang slowing down  and overhang cooling feature of BambuStudio, so we can't apply it for wall at the moment.
When arachne is applied to walls, the line width of wall may be various rather than fixed. And then it's difficult to calculate how much percentage of width of the wall extrusion line is unsupported.

We are still try to fix this confliction.

psiberfunk, 2022-12-15:
It seems like even in the latest version this isn't really fixed.. Arachne in 1.4 still conflicts with slowing down for overhangs.  It's not clear why you shouldn't be able to do both ?

SaltWei, 2022-12-16:
`

> but the slicer still clearly detects overhangs in the sliced view at some threshold. Why should it not be possible to slow down for these overhangs if they can be detected?

Do you mean the overhang wall which colored in blue? The speed of overhang wall still is slowed-down by using bridge speed because it's 100% percentage overhang.
[image]

100% overhang is very easy to be detected. But for not 100% overhang, it's hard to calculate it's overhang degree which expressed by percentage of line width, because the line width is not fixed when arachne is enabled.

SaltWei, 2022-12-16:
Calculate overhang degree of one extrusion line in 2D space is very slow. Pay attention that we are not discussing the angle of overhang mesh.
[image]
(Picture from video of https://wiki.bambulab.com/en/software/bambu-studio/slow-down-for-overhang)

We will try to calculate overhang degree of extrusion line combined with 3D mesh to make it work when extrusion line width is variable, and it will be implemented in further version.

But for arachne, it's now fully enabled and same with behaviour of Prusa.

psiberfunk, 2022-12-16:
> `
>
>
> Do you mean the overhang wall which colored in blue? The speed of overhang wall still is slowed-down by using bridge speed because it's 100% percentage overhang. [image]
>
> 100% overhang is very [...]

Ah , but the line width IS fixed for each line segment .. there are just many more of them .  Is it fair to say that this would significantly slow down slicing ?  I could imagine that those of us with fast multi-core machines might be willing to pay the computation cost . Especially if you could draw the lines in the slicer , then compute the overhang for each segment in parallel with the prior layer .. the problem should scale well.

It might be reasonable to consider telling the user that “this combination requires more slicing power , please know that this will cause slower slicing !”.  At least that would seem like a good warning to have ?

SaltWei, 2022-12-16:
OK. I think maybe I need to describe how to calculate overhang degree of one extrusion line in 2D space, then maybe you can understand how slow it will be when line width is varible.

For a segment of line with specific width,
1 Generate a serial of width value according to how many segments of overhang degree is devided. For us, it's [0%, 10%], (10%, 25%), [25%, 50%), [50%, 75%), [75%, 100%), and 100%。 Six segments of overhang degree.

2 Generate a serial of expolygons by offset lower layer with the serials of width value which generated in step1.

3 Use these expolygons to clip the segment of line, and devide the line into many segment, and every segment is marked with the specific overhang degree.

So you can see, when line width is not fixed, as long as the line width is changed, the serial of width value in step 1 need to be re-generated again. The expolygons of step 2 also need to be re-generated again as well.

When line width is fixed, the expolygon serials only need to be offseted and calculated once. But now, it need to be calculated as long as line width is changed. It's very very slow even model is simple but line width is changed.

We decided to enable arachne first but disable overhang slowing down when it enabled.  People can still benefit from this, since a log of model with details can be printed well without overhang slowing down.

SaltWei, 2022-12-16:
> the problem should scale well by threading

Threading is useless when CPU is 100% already when generating wall.

SaltWei, 2022-12-16:
We will keep improving and try to improve the performance.
But doesn't matter, people can enjoy arachne first and don't need to wait.

YifanWuBambu, 2022-12-16:
This is included in v1.4.

closed 2022-12-16 by YifanWuBambu

psiberfunk, 2022-12-16:
> We will keep improving and try to improve the performance.
> But doesn't matter, people can enjoy arachne first and don't need to wait.

Thanks for the details ! You made the right call.   Should we open a new ticket for this corner case of having both ?

SaltWei, 2022-12-16:
> Should we open a new ticket for this corner case of having both ?

Marked by issue #854