bambulab/BambuStudio #470: [Feature Request] Nozzle Wipe Button

Draakul69, 2022-10-20:
Please add a button into bambu studio and bambu handy to clean with nozzle with the cutter tool outside of a print. This would be useful as sometimes there's a buildup of filament on the nozzle tip that could affect auto bed levelling.

RobinWhitfield, 2022-10-20:
Immediately before ABL, the nozzle cools down, wipes, and then moves to the exposed steel bit of the bed before cooling down to 140C and slowly rubbing any excess material completely off the tip of the nozzle.

Therefore I am struggling to understand how another option for a wipe would help in this instance.

Can you clarify the issue you're actually having, or is this a theoretical concern?

Draakul69, 2022-10-20:
I am closing this as I have realised the wipe is executed after resuming a paused print

closed 2022-10-20 by Draakul69