bambulab/BambuStudio #471: Allow stacking in Bambu Studio

ptsao88, 2022-10-20:
Currently Bambu Studio does not allow moving a part up in the Z axis.  It will always snap back to the build plate.

There are some scenarios where it may be useful to stack parts and have interface supports between them.

For example, printing many short flat parts.  Normally this would require swapping build plates every hour or two, but if the ability to stack is available, one could build more parts on one build, increasing productivity.

Thanks!

YifanWuBambu, 2022-10-21:
@ptsao88 
You can merge several objects as a whole one. Then each original object becomes a part of the result object.
It is allowed t
[image]
o stack parts inside the object.

closed 2022-10-21 by YifanWuBambu

ptsao88, 2022-10-21:
Perfect thank you @YifanWuBambu!  You support team was not aware of this.  Cheers.