bambulab/BambuStudio #474: arrange all objects broken

theastadler, 2022-10-20:
[image]
[image]

Since the most recent update, when using the "arrange all objects" function, some items are placed on top of each other.

manch1n, 2022-10-21:
@theastadler Could you please share the source file to help us debug it?

OmiCron07, 2022-10-21:
Totally broken for me:

[image]

[Gizo the Spider.zip](https://github.com/bambulab/BambuStudio/files/9841395/Gizo.the.Spider.zip)

tangketan, 2022-10-31:
@OmiCron07 Sorry that I can't reproduce the bug. It's OK on my side. Version is the same as yours, 1.2.0.9.
[image]

OmiCron07, 2022-11-01:
I can reproduce the problem with the included file and the version '01.03.00.12'. I even have another project with the same problem:

[image]
[image]

SaltWei, 2023-03-02:
v1.5.0 public beta has been released. Please let us known if still exist.
Thanks!

closed 2023-03-02 by SaltWei