bambulab/BambuStudio #475: Ironing is broken

OmiCron07, 2022-10-21:
Used a modifier to iron a part and it does not print the layer on which it does the ironing.

[image]

There is the project file: https://filetransfer.io/data-package/Rljajrdf#link

OmiCron07, 2022-10-21:
Even with a simple project with a cube:

[image]

OmiCron07, 2022-10-21:
Found out that the bug is only in the latest version (01.03.00.12).

SaltWei, 2022-10-21:
Choosing monotonic pattern for top surface can be a workaround

SaltWei, 2022-10-24:
Fixed patch will be included in next version.

YifanWuBambu, 2022-12-16:
This is fixed in v1.4.

closed 2022-12-16 by YifanWuBambu