bambulab/BambuStudio #476: Settings By Layer Height [Feature Request]

rwoldberg, 2022-10-21:
I would like the ability to change the settings based on layer height.  At least one other slicer that I have used has this feature and there are times when it is extremely useful.  One example is when there is a section of a part that you want solid infill but not the rest.

YifanWuBambu, 2022-10-21:
@rwoldberg 
Please try the cubic modifier, which could also do the same thing.

rwoldberg, 2022-10-21:
That works, and I see how it is more flexible than just settings by layer height, but it is not as easy either.

psiberfunk, 2022-10-25:
The cubic modifier is definitely less intuitive but more flexible.  Since Bambu strives to be approachable, however, it would be good to have a by-layer-height thing.. particularly because in some cases people may want to change properties gradually over layer height for cosmetic reasons.

halofx, 2023-01-08:
YES!!! Please implement Height Range Modifiers, as in Prusa Slicer. I desperately miss this function. There are several differences between height ranges and block modifiers. First is clutter, adding all those blocks to the screen can clutter up the view, and make it difficult to make additional changes. Second is precision, It is much easier to target the exact layers where changes should happen with a height range, than blocks. Specially considering that the block values are relative to the base model, so the math can be much more complex, and counter intuitive.

liftbag, 2023-02-23:
> YES!!! Please implement Height Range Modifiers, as in Prusa Slicer. I desperately miss this function. There are several differences between height ranges and block modifiers. First is clutter, [...]

I totally agree with you.

psy0rz, 2023-04-25:
closed in 1.6.0

closed 2023-07-21 by SaltWei

mduft, 2024-06-06:
Closed as complete means this feature exists...? How can I use it?

SaltWei, 2024-06-07:
> Closed as complete means this feature exists...? How can I use it?

[image]
[image]

mduft, 2024-06-07:
That is... perfect! Thanks :)