bambulab/BambuStudio #480: UI, when clicking on Color Painting your model will be centered automatically

maxmax3, 2022-10-21:
UI Prepare: When you move your (zoomed) model out of center to a certain view and then choose the tool Color painting, your object is centered again automatically. That's a little bit annoying, because you have to move your object again to the view, you chose before.

YifanWuBambu, 2022-10-21:
@maxmax3 
We may optimize the view behavior like this.
1. when entering color painting tool, auto center and zoom currently selected object for easy painting
2. when leaving color painting tool, restore to the same view that is before entering the color painting

kurris19, 2022-10-21:
In my opinion, those suggestions would not be an improvement.  When I'm painting support blockers, for example, I flip between the preview (where I zoom in on trouble areas), then back to prepare and into support painting.  I want and need the view to be exactly the same as preview and prepare, so I can target the area of interest and not have to zoom/pan to find it again.

Color painting should probably work the same way so as to be consistent across the UI, which is beneficial to the user.

YifanWuBambu, 2022-10-22:
Thanks. I know your actual senario. Will change the behavior.