bambulab/BambuStudio #481: pc hang when slicing

PowerChaos, 2022-10-21:
hi,
this is just a general bug report related to the bambu studio

for some reason when i am slicing a intensive file ( like 2 days to print, hybrid mode  ) it hangs my complete pc when it is at 65%
it prevents internet from working , prevents opening task manager and let chrome crash

but nothing is crashed as i see the slicer slowly going to the next slicing part ( from layer 521 to layer 652 for example ) 

this is easy fixed if i just change the affinity to only 6 cores instead 8 cores
can you check this please ,as it it not fun that all programs crashes because bambuslicer want to dominate everything and even let crash task manager .. 

Thank you
Greetings from PowerChaos

ChrisG-hub, 2022-10-21:
makeuseof.com
https://www.makeuseof.com › how-...
How to Manually Allocate CPU Cores to a Program on Windows 10

that will help you.

PowerChaos, 2022-10-21:
Thank you
and yes it could help (and i need to use it )  , but the program should not do that in first place
it should not take priority over task manager and certainly not crash your internet and everything else because it like to use 70% cpu

i got other programs running ( like fusion 360 ) that can use 100% cpu and still let windows take priority
but this is the only program i know off that let everything crash as it want to bump up its own priority to the top.

with crash i mean that it crash everything except his own program , a thing a virus would do and that no program should ever need to do ( same as memory leaks , just assign a fixed memory value it can use  to bypass the memory leak ?) 

Greetings from PowerChaos

lanewei120, 2022-10-28:
> Thank you and yes it could help (and i need to use it ) , but the program should not do that in first place it should not take priority over task manager and certainly not crash your internet and [...]

Hello

thanks for report this issue,
Could you share your 3mf for us to have a debug?

PowerChaos, 2022-10-28:
as requested is here a extreme example file
the file that i tested it on can i not share as it is licensed to hex3D ( 4skull_chalice and the 2 others in the same zip on the same plate makes it a big print to calculate supports ) 
but the below file will provide the same result 
mainly at 65% when genererating support
[2_colors_Wall_Street_bull_by_dizingof-_3dizingof.zip](https://github.com/bambulab/BambuStudio/files/9890925/2_colors_Wall_Street_bull_by_dizingof-_3dizingof.zip)

my pc spects
I7 7700K @ 4.5 GHZ
32GB DDR4 Ram
Crucial P1  1TB NVME or SamSung 860 NVME
Windows 11 ( not official supported cpu ) 

Hope this can help to debug
Greetings from PowerChaos

lanewei120, 2022-10-31:
do you mean you could not do other things when the slicing occupies almost 100% of CPU?

I can reproduce the 100% case in support generation
but it doesn't affect my other operations
I can switch to my browser and access the web smoothly
bambu Studio's priority is normal, not other higher values
[image]

we use TBB in the slicing for parallel acceleration
and currently it will occupy all the cpus when needed

the strange thing is that why it blocks your other applications even it occupies a high cpu ratio
maybe we can limit the max cpu cores it can be used when slicing
we need to check whether it is the common case on others' computers before that
meanwhile we will check the support generation's logic for optimizations

ChrisG-hub, 2022-10-31:
I can't reproduce this, minimum CPU 79%/maximum CPU is 94% totaling 1700MB memory
Core i5, 16MB RAM
All is depending on background programs, AV...
[image]

PowerChaos, 2022-10-31:
When I am home I going see if I can reproduce it 

The thing the slicer does. It generate support and at 65% ( the support generation ) then it freeze the full pc. Including internet. Chrome and discord that are open ( sometimes fusion 360 to )
My ram use around 24gb when the program is around 4gb ram 

I use 3 screens ( 2x 1080p and 1x 4K ) 

I got a dynamic background running. So it plays a movie at the background ( wallpaper ) 
It is the fist one that crashes before others follow and first one that unfreeze 

I also run windows 10 and immunet av. But it does that also before I got those installed 

Same file sliced with prusa slicer 2.5 does not produce the same error.

I think it is indeed the multicore. If you can keep 20% free ( or atleast 1 core ) then this problem does not exist

Maybe it is my windows 11. Maybe a program that like to steal priority ( like nvidea shadowplay ) or the high ram usage ( got 32 gb so should be fine if 24gb is in use ) 

Strange thing is that it only happends when using support generation ( mainly hybrid ) 

If I slice files and later I play a game the games run smooth when they also use 100% cpu ( deathloop ) 
So it is related to the slicer only for some reason :(

PowerChaos, 2022-11-12:
This is the latest i sliced ( [Flexi Phoenix (BETA)](https://www.patreon.com/posts/flexi-phoenix-74541679) at 145% [image]
and at 15% it fully crashed the pc , when i could take the screenshot the pc was back responding

if i slice it at 100% size or other "less intense" files then the pc runs fine
so i guess it would indeed help to limit the amount of cpu it can use , or to limit the parallel method so atleast a single core stays free at all time so windows can still react when needed

sorry for late reply
Greetings from PowerChaos

MusangKingBambulab, 2022-12-09:
> as requested is here a extreme example file the file that i tested it on can i not share as it is licensed to hex3D ( 4skull_chalice and the 2 others in the same zip on the same plate makes it a big [...]

Thanks @PowerChaos , we have noticed this issue and have been continuing to optimize the tree support generation. In your model, I spent ~37 mins to finish the whole slicing with tree support and ~15mins with normal support. Unlike the mechanism of normal support generation which can be generated in parallel, it is hard to apply TBB for the whole procedure (tree is generated layer by layer) but just limited to some specific operations. The configurations of my hardware are listed below for your reference:
CPU: AMD Ryzen 9 5900HX - 8 cores/16 logical processors
RAM: 32GB DDR4
SSD: 1TB SN730
OS: Win 10
BambuStudio Release： v1.3.1.1.