bambulab/BambuStudio #483: If AMS is connected there is no Unload button.

Superbobo75, 2022-10-21:
If AMS is connected there is no Unload button in case filament is loaded via the side spool even though the display indicates it is Loading filament. And the system knows it is not from the AMS.
If I disconnect the cable from the AMS, the Unload button shows back up.

[image]
[image]

YifanWuBambu, 2022-10-22:
@Superbobo75 
When AMS is connected, please go to AMS page to load/unload.

Superbobo75, 2022-10-22:
But since the new Bambu/AMS firmware, there is no need to disconnect the cable to print from the AMS or the side spool. When I have AMS connected but I don't Loading filament from AMS but side spool, there is no Unload button on either screen. 
[image]

There is an indication of loaded filament on the first screen, but on the AMS screen all filaments are unloaded. This means that filament is loaded from the side spool. Despite this, there is no Unload button on either screen.

[image]
[image]

YifanWuBambu, 2022-10-24:
Oh, you are right, We should keep the unload icon. Thanks for reporting the issue.

Superbobo75, 2023-01-19:
Probably one of the most commented bugs at the moment and I posted this here back in October. Is there any way you can move this forward in terms of a solution?

SaltWei, 2023-02-16:
The latest printer firmware has handled this issue. Please update and try.

Studio will also handle this issue in next release version.

SaltWei, 2023-03-01:
FIx is available in v1.5.0 Public Beta

closed 2023-03-01 by SaltWei