bambulab/BambuStudio #484: Wall and Infill print order.  Add   Inner/Infill/Outer

randyre, 2022-10-21:
Seems that this one order is missing.    Inner/Infill/Outer

YifanWuBambu, 2022-10-22:
Hi @randyre ,
In what senario do you need this order?

randyre, 2022-10-22:
I had a large domed shape that I was printing, but it needed the infill to rest on for the outer layer, So wanted to print the inner first. Infill first was not sticking reliably, so wanted to give it at least one wall and it needed to be the inner since the outer needed the infill for support.  It is not a big deal as it is reasonably rare to need this.

YifanWuBambu, 2022-10-24:
"so wanted to give it at least one wall and it needed to be the inner since the outer needed the infill for support"
Could you share the model file to us?

randyre, 2022-11-13:
[BD5 Fuse.zip](https://github.com/bambulab/BambuStudio/files/9998089/BD5.Fuse.zip)

closed 2023-08-29 by randyre