bambulab/BambuStudio #486: Adaptive Layer Height Issue

rwoldberg, 2022-10-22:
It seems that adaptive layer height does not work when a prime tower is enabled.  When the prime tower is off adaptive layer height appears to work as expected.  I have attached a simple example that shows the behavior.
[AdaptiveTest.zip](https://github.com/bambulab/BambuStudio/files/9843257/AdaptiveTest.zip)

YifanWuBambu, 2022-10-22:
@rwoldberg 
Yes. This is is a limitation of slicing program.
Maybe we need to add a prompt message for that.

rwoldberg, 2022-10-22:
It would be nice if it could be modified so that dynamic layer height worked with a prime tower.  Especially in this case where the prime tower is only three layers thick.

Superbobo75, 2022-11-16:
I also agree with the ability to edit adaptive layers like in Prusaslicer. In many cases it instead makes illogically ugly adaptive layers making it unusable.

YifanWuBambu, 2022-12-16:
This is included in v1.4.