bambulab/BambuStudio #487: Ability to set preference for Bed Leveling and Flow Calibration options.

siganberg, 2022-10-22:
Ability to set or save default value for Bed Leveling and Flow Calibration checkboxes. Having it checked as a default is just one thing to forget to uncheck for users who preferred skipping these steps on most prints. 

[image]

YifanWuBambu, 2022-10-27:
@siganberg 
It is strongly suggested to perform bed leveling before every print.
Flow calibration should be performed each time you change the filament, which is not a low frequent operation.
 so we did not provide such an option.

siganberg, 2022-10-27:
@YifanWuBambu, It is just a waste of filament and time for doing bed leveling and flow calibration all the time. For most of the prints, you will stick on one filament for a while and this is a really unnecessary step. 

I've been printing without this, heck I don't even do this even changing filament type, and still works perfectly. 

Just look at the amount of time it saves for me by modifying the start Gcode without these 2 steps. 

[image]

[image]

46 secs vs 6 minutes.

YifanWuBambu, 2022-10-28:
@siganberg 
Okay. We have added this into the plan list.