bambulab/BambuStudio #492: [BUG] Windows 1.3.0.12 When changing presets, reverts slot 1 to Polyterra PLA

MizuUsagi1, 2022-10-22:
Minor bug. When going from one print preset to another, filament slot 1 is automatically changed to Polyterra PLA.

lanewei120, 2022-10-28:
> Minor bug. When going from one print preset to another, filament slot 1 is automatically changed to Polyterra PLA.

Hello

thanks for reporting this issue
Could you share your 3mf or give some picture/gif to help us to reproduce this issue?

lanewei120, 2022-11-21:
@MizuUsagi1 
we could not reproduce this issue

could you help to upload a simple 3mf which has this issue?
or could you give a snapshot before you switch the print?

SaltWei, 2023-10-08:
Too old bug issue and no update from user for a long time. Close.
Reopen if has new response or update.

closed 2023-10-08 by SaltWei