bambulab/BambuStudio #494: Top/bottom layer direction option

randyre, 2022-10-22:
I've been just changing the model orientation to achieve the line direction needed for the top and or bottom layers.  But now I have a model that's too large and I can't rotate it and realize that I can't change the line direction for the top layer.

SaltWei, 2022-12-08:
> I've been just changing the model orientation to achieve the line direction needed for the top and or bottom layers. But now I have a model that's too large and I can't rotate it and realize that I can't change the line direction for the top layer.

The infill direction also influence the direction of top and bottom layer. For this case, you can set infill direction directly.

SaltWei, 2022-12-08:
> Secondly, top and bottom options such as direction and pattern should be under top/bottom shells, not under infill.

Agree. Will move it.

closed 2023-02-03 by SaltWei