bambulab/BambuStudio #495: [Bug] User filament profiles changing when system profiles are updated.

RobinWhitfield, 2022-10-22:
When Bambu Studio was updated from 01.02 to 01.03, various system filament profiles had their bed temperatures adjusted.

That's fine, system profiles can be optimised over time.

What's not fine is that a load of my *user* profiles have also automatically had their bed temperatures changed.

Has anything else been automatically changed?

As you can't create a filament profile in the GUI without modifying an existing profile, there should not be any kind of profile inheritance happening.

RobinWhitfield, 2022-10-22:
Custom filament profile - even when the filament type is changed, it continues to inherit settings from whatever you were initially looking at when you created it.

[image]

SaltWei, 2022-10-27:
> What's not fine is that a load of my _user_ profiles have also automatically had their bed temperatures changed.

This is reasonable, because we need to update system preset for improving. User profiles is saved based on system preset and loaded according to difference only.

If you want user_ profiles unchanged when system preset is changed, we can provide a option to break the inheritance relationship between user profile and system profile. But then the user profile can't be uploaded to cloup then.

RobinWhitfield, 2022-10-31:
An option to remove inheritance would resolve most of the issues, but firstly I think inheritance needs to be made clear in the GUI so a user is aware of exactly what is happening. Ideally this would show exactly which values are being inherited, and which are not.

When a new version of Bambu Studio is released, I'd also like to see any changes to filament profiles be clearly identified to a user, as fundamentally this affects how parts print, and what you think of as improvements may end up negatively impacting how someone is using their printer.

The bed temperature change is not a particularly serious change, but in my case, the higher bed temp noticeably increased elephants foot with one of my filaments (as the bed temp is now higher than its glass transistion temperature), and I was having zero issues with bed adhesion previously so this change did not benefit me in any way.