bambulab/BambuStudio #496: Feature Request: Set Number of Instances/Fill bed with instances

harriettubgirl, 2022-10-22:
Hello I just wanted to put in my request for re-adding the ability to add instances or set number of instances of a specific model from prusaslicer/superslicer as it's implemented over there. 

I find it's far easier to work with than using clone.

Thanks!

YifanWuBambu, 2022-10-27:
@harriettubgirl 
Bambu Studio optimized the clone logic.
All cloned objects shared the same geometry data and only one object is really sliced.

rncfiles, 2022-12-15:
I agree instances is important functionality.  The key benefit of instances is that they inherit the object print settings.  Clones, once created must be managed as separate objects.  What is interesting is that BambuStudio appears to currently support instance functionality.  If a 3mf file that was created in PrusaSlicer has model instances, they will carry over when imported to BambuStudio.  They show as "instance" and behave as expected.  It just seems like in BambuStudio, the + and - options were just omitted.  

Please add this functionality.

jeffamelang, 2022-12-22:
Cloning and instances are fundamentally different. I don't think that the end user cares that "only one object is really sliced", we would like to have a primary copy of the geometry and a bunch of instances, so that if you make changes to the primary copy, that change affects all of the instances. 

Please add this functionality.

harriettubgirl, 2022-12-22:
Thank you for adding your feedback on this! I agree, it has a lot of important functionality and I really hope it's added!

ftheriachab, 2023-01-07:
Definitely would like this feature implemented.  When batching out parts, it's so nice to have the slicer determine the maximum number of objects in can print.

kozakm00, 2023-01-13:
Agree with OP, need to add instancing to the Bambu studio. Prusa slicer has it

pennystockss, 2023-02-07:
This is huge for me. Printing 3-20 versions of the same object routinely. Kind of a pain having to delete every one but a single one. Make a change, then clone it back.

khibchen, 2023-03-01:
Linked instances that all share the same parameters and modifiers is a critical feature for anyone that produces multiple identical parts on a build plate. The way it’s implemented in Prusa Slicer is very powerful and simple to use. The clone function in Bamboo Studio basically just makes ‘dumb’ copies of a part. Any changes to one part are not automatically reflected in the other parts.

The absence of linked instances in Bamboo Studio requires a lot of tedious effort to make even small changes to multiple identical parts. I routinely print 120 small parts on a build plate. For various reasons it’s sometimes required to change the parameters, a modifier or some other detail of that part. In Bamboo Studio I must delete all instances except one, make the changes to the one part and then re-create the desired number of clones. Finally, I must redo the arrangement of the parts on the build plate.

Another problem with the ‘clone’ approach is that there is no easy way to know how many objects are on the build plate. You must count them! In Prusa Slicer, under each part there is an ‘Instances’ item that can be opened and you can see all the instances listed and clearly numbered. If you want to change the number of instances in Prusa Slicer, you just right-click on an instance and select ‘Set number of instances…’. There are a lot of things I like about Bamboo Studio and the X-1 Carbon an absolutely awesome 3D printer. Bamboo Labs seem to be very responsive to user feedback. I hope they will add this feature soon.

harriettubgirl, 2023-03-01:
> Linked instances that all share the same parameters and modifiers is a critical feature for anyone that produces multiple identical parts on a build plate. The way it’s implemented in Prusa Slicer [...]

Thanks for your reply to this! I just posted it on the forums to hopefully get more dev attention if you wouldn't mind throwing this reply or showing support over there! Much appreciated

rncfiles, 2023-03-01:
I think it might help to change the title and original ask.  The way it reads, the request is for the fill bed feature.  I think this should be secondary feature request with the priority goal of adding the 'instance' construct.  Clones and Assemblies to address the need.  How instances are implemented in PrusaSlicer is exactly what is needed and should be straightforward.

@YifanWuBambu since you have been the only Bambu rep to comment, can you please help to get a dev assignment on this?

KD7LRJ, 2023-03-02:
Yes, please implement the ability to edit a single object and have all copies on the plate inherit the settings.

I'm currently working on producing 200K of these little game pieces with 6 printers (so far) and this would have been extremely helpful while developing the custom settings for them. All of the things a previous poster mentioned applied to my experience as well. The worst was having to recreate the custom layout every time I made an update to the settings. I went through 88 revisions (I renumber each one before I send it to the printer) over the past 3 months perfecting the process and am about half way through the 400+ rolls of PETG now. Printers are doing great, btw!

[image]

YifanWuBambu, 2023-03-02:
Instance mechanism is not supported in Bambu Studio. We may implement some of the requirements:
1. fill a bed plate with a selected objects and auto rename the copied objects with a suffix, such as cube-1, cube-2, etc
2. support to show the count of selected objects. In this way, you can select a range of objects by rectangular selection(shift+left mouse drag) and see the total number at the prompt message

harriettubgirl, 2023-03-02:
> 

I greatly appreciate your response but I do think you guys are overlooking a powerful feature by simplifying it with the clone mechanism! Instances can easily serve the same function, but with many added benefits.

pennystockss, 2023-03-02:
That is very unfortunate to hear. Here's a video of the flow that must be used with the cloning tool. Watch the video and please understand our frustration. Every single time there is a minor change, we have to do this. It doesn't happen often but once you get everything set and ready to go then spot a tiny little defect or change, you have to start over.

Honestly the only thing that would work without instances is if the auto arrange tool was smarter. It just looks at the bounding box around the object and doesn't actually take the real object into consideration. We might have 30 things on the bed. With auto arrange we can fit half but with a "smart" auto arrange and not having to auto position and rotate, we can fit the full 30 without manually having to arrange. This video perfect example, we can clearly print 3 at a time and save a ton of purge waste. But auto arrange thinks we can fit only 1.

https://user-images.githubusercontent.com/60720749/222517325-24a4eab4-ca85-4781-9e36-aa19d7ef3fda.mp4

kozakm00, 2023-03-03:
I couldn't agree more that instancing feature is desperately needed and I'd not like to turn this reaquest to "fill plate" only.
Please refer to Prusa slicer where instancing is super convenient: 
[image]

faroutride, 2023-03-05:
+1! 
- INSTANCES (where all children inherit the exact same parameters of the parent) are very important for "mass" production. 
- CLONES are a pain to deal with for mass production, very time consuming and prone to mistakes.

Please just do the same as PrusaSlicer! Thanks! :)

pennystockss, 2023-03-07:
Here is another great example to the developers why it's extremely needed. Look at this picture. They want half red/blue and half blue red. I got 1 plate perfectly aligned, copied to another, because auto align will only put 2/3 on the plate. Now I need to change one plate to blue instead of red. Do I go one by one and change the colors of each and every object or delete, clone and manually realign? Either way it's a pain. Also I noticed the text on the first batch could could be better, so I want to slow the speed of text down to like 25mm/s. Do I literally click on each one and adjust?? Serious pain in the rear!
[image]

RyanEwen, 2023-03-09:
I currently have to delete every copy and then recreate and reposition each one of them which drives me insane. +1 for this feature back.

MadMax389, 2023-03-21:
+1 again for all the reasons mentioned above.

vtjballeng, 2023-05-15:
This is really important for my models as well. I'd like them:
- In the same orientation (inherinting the parents attributes).
- To have an instance count so I quickly know how many I am producing at a glance. 
- To have a +1 instance button.
- To have a feature to fill the bed with instances. 

For example: How many are here? Do I need to count them manually? 
[image]

faroutride, 2023-08-04:
"Fill bed with copies" feature was added in Bambu Studio 1.7, however "Instance" feature is still missing.

Per PrusaSlicer documentation (https://help.prusa3d.com/article/copy-paste-instances_1769#object-instances):
"Objects instances are essentially copies of an object, which are **linked together and share the same settings and orientation**, except for the Z-axis rotation. They’re **extremely useful when you want to quickly set-up a print with multiple copies of the same model**.
When you scale any of the instances, all of them will change in scale. When you cut the bottom of a model, all of the instances will be cut, etc."

Please consider introducing "Instances" to Bambu Studio.

RyanEwen, 2023-08-04:
The lack of this feature is really slowing me down when I update my models each iteration. It's utterly painful. One model of mine consists of 5 objects and each requires different slicing options and I print 12 of them on a bed at once.

Currently to itetrate, I have to delete the 11 copies, replace the original objects, copy and paste 11 times, and then painfully lay the copies out as they are not automatically placed well on the bed. 

If I could leave the 11 copies placed and just replace the original objects, all 12 would instances would get replaced at the same time without extra effort.

valentt, 2023-08-05:
Everyone who uses Bambu machines for mass production needs this feature, this is a huge oversight. And no other feature can replace this feature. Is it possible to add this feature via Orca Slicer or it is a fundamental issue that only Bambu developers can fix?

pennystockss, 2023-08-06:
Not to mention the size difference. Here are a few screenshots. One is of my Print folder, sorted by size. One is actually over a gigabyte! lol. If you have a few plates full with STL's, it bogs Bambu Studio down soooo bad.

Other 2 screenshots are PrusaSlicer and BambuStudio with 30 Flexi Astronauts on them. Granted I'd never print this way but to show how big of a difference it is. But 13mb vs 399mb for exact same plate 🤔

![Size](https://github.com/bambulab/BambuStudio/assets/60720749/6e49
[image]
[image]
4067-30f7-44dd-9f81-f4f1b89384f4)

MadMax389, 2023-08-06:
I wonder if this issue is getting reviewed by BL. It's been open almost a year and no one has been assigned. There are other similar requests open like #1817 and #1766. In #1766 @tangketan thinks the implementation is too difficult and suggests the feature is useless, but is willing to discuss. The code base exists in PS, but I'm no coding expert so I wouldn't know how difficult it would be to integrate into Studio. This feature is the one I miss the most for all the reasons stated above.

harriettubgirl, 2023-08-07:
> Cloning and instances are fundamentally different. I don't think that the end user cares that "only one object is really sliced", we would like to have a primary copy of the geometry and a bunch of [...]

I really hope so! I'm still watching this request closely :) I also still believe instance objects as it's done in prusa slicer is far more powerful than it's replaced clone function!

julie777, 2023-11-17:
This is a very important feature. The lack of instances causes issues when a project has multiple plates, many objects and some objects being copied many times on a plate. **The object list is huge!** Managing settings for all the clones is problematic. With true instances, the object would only be shown once in the object list along with an optional indicator of the number of copies. Selecting any of the instances should select the actual object for settings modification.

Of course, my request to be able to apply print profiles to individual plates would also help with this issue.

black8592, 2023-12-27:
I made an object with text in Fusion360, filled the plate and now i want to swap the STL, same object different text. I would have expected that clones follow the original object but due to lack  of instances this is not possible for now. Fill plate is also buggy, create a shap 40x55mm in top left, fill plate and see the right 3 rows beeing filled and the few left rows are not (k1 max build plate)

faroutride, 2024-01-23:
Bump! 
Still waiting for instances to be implemented...

RozeDoyanawa, 2024-05-24:
I would also really like an instance feature. 
Ponder this, you import an object, and add text or primitives to it in the slicer, if you clone it and want to change something, you will have to remove all objects, do your changes, and re-clone them. This means you have to redo the positions on all items if you manually moved them around. 

With instances, you can make changes to any of the instances and all get immediately updated, no need to re-position or otherwise.

For example, I made boxes with numbers engraved recently, I added primitives in the recesses in the slicer to fill the numbering with another color, works fine, but when I have 9 or 10 of them on the bed, each carefully positioned to fit, Adjusting the up to 35 disks on each clone is quite tedious. 

Please implement an instance feature!

morganvenable, 2024-07-08:
Still missing, still a disaster for production and an obvious gap with the Prusa ecosystem.  If I can get a reasonable machine that prints anywhere near as nice as my X1C fleet, I will switch back to PrusaSlicer just for this -- it's murder on my production workflow.

rtm7777, 2024-08-13:
Tried this slicer after years of using SuperSlicer, I'm making mostly mass production parts, for half an hour I was thinking I just can't find proper button, this is so sad.....

kozakm00, 2024-10-17:
How come this is still being ignored?

EricDimitri, 2024-10-22:
Not much to add. This is a must, and will be a great option to have soon. Thanks!!!

ignacio-oliver, 2024-10-22:
+1
Please add this functionality. It would help a lot in batch editing/printing.

dimitarsp, 2024-12-01:
I am just here to add my voice to the need of instancing, where settings adjustments of one object get applied to all instance. Must have feature!!

KhalidLR, 2025-01-21:
Same! Please add this feature!

faroutride, 2025-12-30:
Still waiting for instances to be added to Bambu Studio...