bambulab/BambuStudio #497: [Enhancement] Per build plate options

bgreat, 2022-10-22:
Allow setting of slicing options that are unique per build plate.

- Quality, Layer height/Line width/Ironing/etc.
- Strength, Walls/Shells/Infill/etc.
- Speed
- Support
- Others, Bed adhesion/**Print sequence**/etc.

This would allow for post assembled multipart models to have all printing set up in a single project file with the most efficient settings for each plate of the build. This avoids having multiple separate project files for a single complete project.

Thank you.

YifanWuBambu, 2022-10-27:
@bgreat 
We will allow setting bed type (cool plate, high temp plate, etc) for each plate in a recent release.

For your requirement, I think you can use the object setting.
https://wiki.bambulab.com/en/software/bambu-studio/set-parameters-for-selected-objects

bgreat, 2022-10-28:
Bambu Studio currently allows setting the Bed type for a project, but this applies globally to all of the Plates added for the project. My suggestion is to allow settings per Project "Plate". This is very separate from he object settings which are much more limited, though they have been expanded with more of the options.

The simplified version of the request is to make all project settings overridable on a per Plate basis.

This establishes a hierarchy of option inheritance that can be overridden of Project->Plate->Assembly->Object. Rather than creating separate projects comprising one final build, everything could be encompassed in a single project.

As an example, in the image (not the best example, but was loaded):
    * Plate 01, 02 - I would like to print with a Print sequence of By layer, without Timelapse, with a First layer infill speed of 50 mm/s
    * Plate 03, 04 - I would like to print with a Print sequence of By object, with Timelapse, with a First layer infill speed of 105 mm/s
    * 
[image]

I'll k

YifanWuBambu, 2022-12-03:
@bgreat 
We will firstly add a per-plate setting "bed_type" in the next release.

YifanWuBambu, 2022-12-05:
> This establishes a hierarchy of option inheritance that can be overridden of Project->Plate->Assembly->Object. Rather than creating separate projects comprising one final build, everything could be encompassed in a single project.

This is in our plan. We will first add "bed_type" settings for each plate.

halofx, 2023-01-08:
YES! Per plate global settings. I am trying to setup plates where one plate is in vase mode, and another has parts that aren't in vase mode, and Bambu Studio will not allow it. All settings should be available to override at the plate level.

OpenJeDi, 2023-02-25:
In my case it is the print sequence that makes me having to save in multiple projects